intel/qatlib #3: Can't compile with Clang 12: undefined symbol: __sync_val_compare_and_swap_16

AngryPenguinPL, 2021-06-02:
When compile with Clang 12 I see this build error:

```
ld: error: undefined symbol: __sync_val_compare_and_swap_16
DEBUG util.py:634:  >>> referenced by lac_lock_free_stack.h:113 (quickassist/lookaside/access_layer/src/common/utils/lac_lock_free_stack.h:113)
DEBUG util.py:634:  >>>               lto.tmp:(cpaDcChainPerformOp)
DEBUG util.py:634:  >>> referenced by lac_lock_free_stack.h:113 (quickassist/lookaside/access_layer/src/common/utils/lac_lock_free_stack.h:113)
DEBUG util.py:634:  >>>               lto.tmp:(cpaDcChainPerformOp)
DEBUG util.py:634:  >>> referenced by lac_lock_free_stack.h:113 (quickassist/lookaside/access_layer/src/common/utils/lac_lock_free_stack.h:113)
DEBUG util.py:634:  >>>               lto.tmp:(cpaDcChainPerformOp)
DEBUG util.py:634:  >>> referenced 28 more times
DEBUG util.py:634:  clang-12: error: linker command failed with exit code 1 (use -v to see invocation)
DEBUG util.py:634:  make[1]: *** [Makefile:2295: libqat.la] Error 1
DEBUG util.py:634:  make: *** [Makefile:1870: all] Error 2
```

Full build log: https://file-store.openmandriva.org/api/v1/file_stores/1d5f712b01c4e98110a3544c1ebc9fe8b5922d11.log?show=true

GCC comiles fine.

jdschuet, 2021-06-07:
Thank-you for raising this issue.  We have logged ticket internally and the issue will be investigated.  Once we have additional details on resolution and target release date, this ticket will be updated.

gcabiddu, 2021-11-10:
This should be fixed since 21.08. @AngryPenguinPL, can you verify? Thanks!

AngryPenguinPL, 2021-11-12:
@gcabiddu yes, 21.11 compiled with Clang 13.0.0 without problems. It looks like the case is resolved.
Thank you.

closed 2021-11-12 by AngryPenguinPL