intel/qatlib #14: make rpm fails due to missing SECURITY.md

naoyuki-mori, 2022-10-20:
SECURITY.md is not included in dist tar ball.

```
$ make rpm
...
RPM build errors:
    File not found: /home/foo/src/rpm/BUILDROOT/qatlib-22.07.1-1.fc36.x86_64/usr/share/doc/qatlib/SECURITY.md
```

Proposed patch below fixes issue.

```
diff --git a/Makefile.am b/Makefile.am
index c9ea85f..7c1fe95 100644
--- a/Makefile.am
+++ b/Makefile.am
@@ -301,6 +301,7 @@ endif
 dist_man_MANS = qat_init.sh.8 qatmgr.8
 
 EXTRA_DIST = LICENSE \
+       SECURITY.md \
        autogen.sh \
        quickassist/lookaside/firmware/include \
        quickassist/include \

```

gcabiddu, 2022-10-20:
We could just revert the update to the spec and avoid adding including README.md and SECURITY.md.

naoyuki-mori, 2022-10-20:
Thanks for taking this, @gcabiddu 

Not in hurry, so put changes into main for now then release at next cycle would be fine (, if it won't block Fedora upstream review process.)

gcabiddu, 2022-10-20:
Done. We updated the release.

closed 2022-10-20 by gcabiddu