intel/qatlib #29: cpa_sample_code: No device found

kin-cho, 2023-01-20:
Hi, I used to be able to run `cpa_sample_code` successfully, but it's failing now with newer CPUs.

```
bash-4.4# cpa_sample_code
qaeMemInit started
No device found
ADF_UIO_PROXY err: icp_adf_userProcessToStart: Failed to start SSL
quickassist/lookaside/access_layer/src/sample_code/performance/cpa_sample_code_main.c, main():449 Could not start sal for user space
bash-4.4#
```
but there are supported devices:
`bash-4.4# dmesg|grep 4xxx
[   20.609116] 4xxx 0000:6b:00.0: enabling device (0140 -> 0142)
[   20.745647] 4xxx 0000:6b:00.0: qat_dev0 started 9 acceleration engines
...
`
I'm using 22.07.0, and same version of qatlib.

gcabiddu, 2023-01-20:
Hi @kin-cho, what kernel version and distribution are you using?
Did you install by any chance also the out-of-tree kernel driver in your system?

Thanks,

-- 
Giovanni

kin-cho, 2023-01-20:
Hi @gcabiddu, it turned out this was an installation issue, `cpa_sample_code` works fine after a clean install.

Thanks,

Kin

closed 2023-01-20 by kin-cho