intel/qatlib #35: Support for untrusted users

DemiMarie, 2023-03-01:
Currently, only trusted users and VMs can be allowed to use QAT due to lack of parameter checking in the hardware/firmware.  This means that cloud providers cannot expose QAT to their customers, who are not trusted.

One workaround would be to proxy requests thorugh a trusted driver, either in userspace or in the kernel.  `io_uring` could be used to mitigate the performance overhead of context switching.

fionatrahe, 2023-03-08:
Hi DemiMarie,

In qat 4xxx (QAT in Sapphire Rapids), FW/HW has been enhanced to perform param checking.

Also users/processes in one VM are protected from users/processes in another VM. The VM boundary cannot be crossed.
Even within a VM there is some protection, e.g. there is DMA protection between processes.
However, as within a VM there is not complete protection, users in a VM should trust each other and this is enforced by requiring that the users are added into the qat group.

User-space processes can also use the QAT device via the Linux AF_ALGs interface, which passes the requests to the trusted qat driver in kernel space.
However this is a less performant path than the direct access path provided by qatlib.

Can your cloud provider expose QAT to their customers via VMs?
Can you expand a bit more on your use-case and requirements please?

DemiMarie, 2023-03-08:
> However, as within a VM there is not complete protection, users in a VM should trust each other and this is enforced by requiring that the users are added into the qat group.

What is the reason for this?

> Can your cloud provider expose QAT to their customers via VMs?

“This means that cloud providers cannot expose QAT to their customers, who are not trusted.“ was just an example.

> Can you expand a bit more on your use-case and requirements please?

I don’t have a use-case _myself_, but as a security researcher I don’t like people having to punch holes in system security so that their web server can use QAT hardware acceleration.

fionatrahe, 2024-01-08:
@DemiMarie ,
Thank you for bringing up your concerns regarding the security model of Intel's QuickAssist Technology (QAT) and its implications for untrusted users. We understand the importance of security in cloud environments and the need for robust solutions that do not compromise system integrity.
As mentioned, the QAT hardware and firmware in the latest generation (QAT 4xxx series) have been enhanced to include parameter checking, and there are protections in place to prevent users and processes in one VM from affecting those in another. However, the security model of qatlib does require that users within the same processing domain trust each other. They need to do this as they get device resources by accessing the group files in /dev/vfio. This trust is administered by adding them to the same QAT group.
The use of QAT via the Linux AF_ALGs interface is a potential solution for untrusted environments, albeit with a performance trade-off. We acknowledge that this may not fully address the concerns you've raised as a security researcher.
Since the issue you've raised pertains to the architectural design and security model of QAT rather than a specific bug or feature request within the qatlib repository, we believe this discussion would be more appropriately addressed in a different forum. We encourage you to reach out through official Intel support channels.
For these reasons, we will be closing this issue here. However, we value your input and hope you will continue to engage with us through the proper channels to further discuss and potentially influence the security aspects of QAT.

closed 2024-01-08 by fionatrahe