intel/qatlib #38: QAT service fails to start

kin-cho, 2023-03-15:
Hi, `qat.service` fails to start on our system:

`Mar 14 09:31:11 localhost.localdomain qatmgr[3408]: Failed qat_mgr_build_data`

The reason is because the `qat_4xxx` module has not finished loading before `qat_init.sh` starts to run.
I suggest change the target from `local-fs.target` to `network-pre.target`,
to give `qat_4xxx` time to finish loading.
With this change, now I see `qat.service` always starts.
https://github.com/intel/qatlib/blob/7cc89fce6b75a4b2c9ba950e85a3b754ec2a7660/quickassist/utilities/service/qat.service.in#L36

fionatrahe, 2023-03-27:
Hi Kin. It's good you found a workaround. We are looking at the implications of making the change proposed as it might impact other deployments.

fionatrahe, 2023-09-08:
https://github.com/intel/qatlib#qate-90845 Fixed in qatlib 23.08

closed 2023-09-08 by fionatrahe