intel/qatlib #40: Question: Can qatlib work with OOT driver

AsaMyth, 2023-04-07:
From the [INSTALL](https://github.com/intel/qatlib/blob/main/INSTALL) page, Linux kernel v5.11+ is required, which doesn't meet my use cases.

There is an OOT driver which by default use UIO/SRIOV VF interface: https://www.intel.com/content/www/us/en/download/765501/intel-quickassist-technology-driver-for-linux-hw-version-2-0.html?wapkw=quickassist

Can qatlib work with OOT driver? I'm trying to do so but seeing some weird issues.

Thanks.

fionatrahe, 2023-09-04:
No, qatlib is designed to work with the in-tree kernel driver. 
The OOT driver uses uio as the user-space interface, while qatlib depends on vfio.

fionatrahe, 2023-09-08:
closing this as answered.

closed 2023-09-08 by fionatrahe