intel/qatlib #44: Is it possible to use multi generation qat device (c62x and 4xxx) together on a single system ?

sumantakollya, 2023-06-12:
Hi , I am trying to use both qat c62x and 4xxx device to offload crypto operations.
> I tried installing the  QAT20.L.0.9.6-00024 driver but it supports only 4xxx.
> I then tried using the c62x.ko and c62xvf.ko file from QAT1.7 driver by replacing the corresponding file in the new driver but that caused symbol errors.
> Is there a way I can use both of them ?

p.s - this is my first time opening a issue , sorry for any mistakes.

jdschuet, 2023-06-12:
Hello,

The QAT2.0 out-of-tree package (QAT20.L.0.9.6) does not currently support QAT1.7 devices.  We would need to use the QAT1.7 driver.   It is available at: https://www.intel.com/content/www/us/en/download/19734.html

The latest QAT2.0 driver is available at: https://www.intel.com/content/www/us/en/download/765501.html

Are you looking to use both qat c62x device and 4xxx on the same system?  Or the ability to have one driver that works with both platforms?

Thanks,
Joel

sumantakollya, 2023-06-12:
I am looking for the ability to use both c62x and 4xxx on the same system.

jdschuet, 2023-11-20:
Please consult with your Intel representative for additional information on this inquiry.  We'd like to ensure we understand the use case.  

Cheers,
Joel

closed 2024-01-08 by fionatrahe