intel/qatlib #46: null return from allocation being dereferenced in memcpy

ColinIanKing, 2023-06-19:
source: ./quickassist/lookaside/access_layer/src/sample_code/functional/dc/stateless_multi_op_checksum_sample/cpa_dc_stateless_multi_op_checksum_sample.c - function compPerformOp:

lines: 296..311:
```

        if (CPA_STATUS_SUCCESS == status)
        {
            status = PHYS_CONTIG_ALLOC(
                &bufferListDstArray[bufferNum].pBuffers->pData, bufferSize);
            bufferListDstArray[bufferNum].pBuffers->dataLenInBytes = bufferSize;
        }
        if (CPA_STATUS_SUCCESS == status)
        {
            status = PHYS_CONTIG_ALLOC(
                &bufferListDstArray2[bufferNum].pBuffers->pData, bufferSize);
            bufferListDstArray2[bufferNum].pBuffers->dataLenInBytes =
                bufferSize;
        }
        memcpy(bufferListSrcArray[bufferNum].pBuffers->pData,
               sampleData + (bufferNum * bufferSize),
               bufferSize);
```

The memcpy call accesses bufferListSrcArray[bufferNum].pBuffers->pData that may be null if the call to PHYS_CONTIG_ALLOC fails, causing a segmentation fault.

fionatrahe, 2023-06-22:
Thank Colin. taking this into our internal team, will include the fix in the next release.

fionatrahe, 2023-09-08:
https://github.com/intel/qatlib#qate-93278 Fixed in qatlib 23.08

closed 2023-09-08 by fionatrahe