intel/qatlib #47: various warnings found when statically analyzing build with clang-15 clang-scan-build

ColinIanKing, 2023-06-22:
Hi, 

Before I package qatlib for Debian and Debian derivatives I'm running qatlib through some static code analysis tools. I ran it through the clang-15 clang scan build tool and it reported some issues. Some are just redundant stores to variables, the rest look like false positives but I thought I should share the report just in case any real bugs have crept into the code. 
[clang-scan-build.log.gz](https://github.com/intel/qatlib/files/11832534/clang-scan-build.log.gz)

Attached is the report as a gz compressed log.

fionatrahe, 2023-06-22:
Thanks! We'll check them out

fionatrahe, 2024-07-15:
These were fixed in qatlib 24.02. 
I'll rerun the checker just to double-check before closing this ticket.

Ahsan-Atta, 2024-12-16:
A few minor hits are still pending which will be addressed in future release. Thus, closing this issue.

closed 2024-12-16 by fionatrahe