intel/qatlib #55: Readme doesn't explain supported devices

pwright, 2023-10-04:
As a newbie, I've no idea what '4xxx (QAT gen 4 devices)' means, can it be expanded?

jdschuet, 2023-11-08:
Hi Paul - We will update the README accordingly.  4xxx (QAT gen 4 devices) correspond to the QAT HW included on [4th Gen Intel(R) Xeon Scalable Processors](https://www.intel.com/content/www/us/en/products/docs/processors/xeon-accelerated/4th-gen-xeon-scalable-processors-product-brief.html?wapkw=4th%20gen%20intel%20xeon%20scalable%20processors)

arcivanov, 2024-02-13:
What packages support C3xxx devices (current production Deverton Atoms)?
There is an absolute derth of information on the subject.

jdschuet, 2024-02-13:
This would be an out-of-tree (OOT) package instead of qatlib and in-tree.  The OOT package is available at:
https://www.intel.com/content/www/us/en/download/19734.html
At this link for the software package, there is links to Release Notes and Getting Started Guide.  Within the Release Notes document list of supported platforms is included (see section 1 Description of Release)

arcivanov, 2024-02-13:
> This would be an out-of-tree (OOT) package instead of qatlib and in-tree. The OOT package is available at: https://www.intel.com/content/www/us/en/download/19734.html At this link for the software [...]

Where would I file bugs for that package? 
```
/home/admin/qat/quickassist/qat/drivers/crypto/qat/qat_common/adf_ctl_drv.c: In function ‘adf_get_dev_node_id’:
/home/admin/qat/quickassist/qat/drivers/crypto/qat/qat_common/adf_ctl_drv.c:674:22: error: ‘struct cpuinfo_x86’ has no member named ‘phys_proc_id’
  674 |                 if (c->phys_proc_id == 0) {
      |                      ^~
```

jdschuet, 2024-02-13:
> Where would I file bugs for that package?
>
> ```
> /home/admin/qat/quickassist/qat/drivers/crypto/qat/qat_common/adf_ctl_drv.c: In function ‘adf_get_dev_node_id’: [...]

Do you have access to Intel Premier Support? This is one place ticket can be raised.  If not, ticket can be raised at:
https://community.intel.com/t5/Intel-QuickAssist-Technology/bd-p/IntelQuickAssistTechnology

Please include the distro and kernel version along with the ticket. This appears to be case of newer (or older) kernel definitions not being supported with current SW release.

arcivanov, 2024-02-13:
I already can submit a patch PR as I fixed the issue in the code. Is there a GH repository where I can submit a PR?

arcivanov, 2024-02-13:
It's the issue in the most recent kernel, namely due to this - https://lore.kernel.org/lkml/20230728120929.778470356@linutronix.de/

arcivanov, 2024-02-13:
lovely :) Oh well, I'll just carry my own patch then...

arcivanov, 2024-02-13:
Just FYI in case it can percolate to the dev team:

```
--- ./quickassist/qat/drivers/crypto/qat/qat_common/adf_ctl_drv.c.bak   2023-12-06 04:30:08.000000000 -0500
+++ ./quickassist/qat/drivers/crypto/qat/qat_common/adf_ctl_drv.c       2024-02-13 17:42:08.511594415 -0500
@@ -671,10 +671,10 @@
                /* if there is only one physical processor don't need
                 * to do any further calculations
                 */
-               if (c->phys_proc_id == 0) {
+               if (c->topo.pkg_id == 0) {
                        node_id = 0;
                } else {
-                       bus_per_cpu = MAX_PCI_BUS / (c->phys_proc_id + 1);
+                       bus_per_cpu = MAX_PCI_BUS / (c->topo.pkg_id + 1);
                        if (bus_per_cpu != 0)
                                node_id = pdev->bus->number / bus_per_cpu;
```

jdschuet, 2024-02-13:
> Just FYI in case it can percolate to the dev team:
>
> ```
> --- ./quickassist/qat/drivers/crypto/qat/qat_common/adf_ctl_drv.c.bak   2023-12-06 04:30:08.000000000 -0500
> +++ [...]

Much appreciated.  I can raise internal ticket with dev team.

jdschuet, 2024-02-16:
> Much appreciated. I can raise internal ticket with dev team.

Update:  Our internal team was aware of this issue and have update in flight.  Fix is targeted for the next release.

arcivanov, 2024-02-16:
Yep, there is another one related to the build:

```patch
--- ./quickassist/build_system/build_files/OS/linux_2.6_kernel_space_rules.mk.bak       2024-02-16 16:22:06.632987112 -0500
+++ ./quickassist/build_system/build_files/OS/linux_2.6_kernel_space_rules.mk   2024-02-13 20:05:42.643062646 -0500
@@ -72,7 +72,7 @@
 
 $(LIB_STATIC): dirs
        @echo 'Creating static library ${LIB_STATIC}'; \
-       $(MAKE) -C $(KERNEL_SOURCE_ROOT)/ M=$(PWD) obj-m="" KBUILD_BUILTIN=1; \
+       $(MAKE) -C $(KERNEL_SOURCE_ROOT)/ M=$(PWD) obj-m="" KBUILD_BUILTIN=1 single-build=1; \
        echo 'Copying outputs';\
        test -f lib.a  &&  (ar -t lib.a | xargs ar -rcsD $(LIB_STATIC)); \
        test -f $(LIB_STATIC)  &&  mv -f $(LIB_STATIC) $($(PROG_ACY)_FINAL_OUTPUT_DIR)/$(LIB_STATIC); \
```

fionatrahe, 2024-04-15:
@pwright The set of devices supported is documented here: 
https://intel.github.io/quickassist/qatlib/requirements.html#supported-devices.

@arcivanov Closing this as the original question has been answered.  You can post any issues relating to the QAT OOT package here:  
 https://community.intel.com/t5/Intel-QuickAssist-Technology/bd-p/IntelQuickAssistTechnology