intel/qatlib #58: allow /etc/sysconfig/qat to have comment lines

mrkz, 2023-10-25:
It's a common practice for configuration files in `/etc/sysconfig` to have comments in them, some examples on my current `Fedora 38` machine are:

```bash
$ cat /etc/sysconfig/firewalld
# firewalld command line args
# possible values: --debug
FIREWALLD_ARGS=
```

and

```bash
$ cat /etc/sysconfig/anaconda
# This file has been generated by the Anaconda Installer 38.23.4

[General]
post_install_tools_disabled = 0
```

however, if I use comments on `/etc/sysconfig/qat` I get failures in configuring the device.

Please allow the `/etc/sysconfig/qat` file to be able to contain comments in them and ignore lines starting with `#`

jdschuet, 2023-11-20:
Can you confirm whether this is observed with comments as shown above?   (where comment starts at the beginning of line)
Or is this observed with comments that occur after the contents of the line.. something like:
POLICY = 1  # This is comment

mrkz, 2023-11-21:
I had both comment patterns, but then I was contacted internally and got confirmation that comment-only lines are handled correctly, so I'd say the culprit may be the later
```sh
POLICY=1 # comment
```

jdschuet, 2023-11-21:
Thank-you for confirming.  I've recreated the issue and also see that it affects comments at the end of line.  I've raised internal ticket to track this through resolution.   Thank-you for raising this issue.

jdschuet, 2024-07-09:
Update with planned resolution.  Comments are supported at beginning of line, but not at end of line.

closed 2024-07-15 by fionatrahe