intel/qatlib #60: install: Job for qat.service failed because the control process exited with error code

kkurzacz-intel, 2023-11-03:
Hello, I have a problem with installation of qatlib. `make` is okay, but when I type `sudo make install`, I get unclear error:

```
$ sudo make install
make[1]: Entering directory '/home/kkurzacz/benchmark/qat/qatlib'
 /usr/bin/mkdir -p '/usr/local/lib'
 /bin/bash ./libtool   --mode=install /usr/bin/install -c   libusdm.la libqat.la '/usr/local/lib'
libtool: install: /usr/bin/install -c .libs/libusdm.so.0.1.0 /usr/local/lib/libusdm.so.0.1.0
libtool: install: (cd /usr/local/lib && { ln -s -f libusdm.so.0.1.0 libusdm.so.0 || { rm -f libusdm.so.0 && ln -s libusdm.so.0.1.0 libusdm.so.0; }; })
libtool: install: (cd /usr/local/lib && { ln -s -f libusdm.so.0.1.0 libusdm.so || { rm -f libusdm.so && ln -s libusdm.so.0.1.0 libusdm.so; }; })
libtool: install: /usr/bin/install -c .libs/libusdm.lai /usr/local/lib/libusdm.la
libtool: warning: relinking 'libqat.la'
libtool: install: (cd /home/kkurzacz/benchmark/qat/qatlib; /bin/bash "/home/kkurzacz/benchmark/qat/qatlib/libtool"  --silent --tag CC --mode=relink gcc -I./quickassist/utilities/libusdm_drv -I./quickassist/utilities/osal/include -I./quickassist/utilities/osal/src/linux/user_space/include -I./quickassist/include -I./quickassist/include/lac -I./quickassist/include/dc -I./quickassist/qat/drivers/crypto/qat/qat_common -I./quickassist/lookaside/access_layer/src/common/compression/include -I./quickassist/lookaside/access_layer/src/common/crypto/sym/include -I./quickassist/lookaside/access_layer/src/common/crypto/asym/include -I./quickassist/lookaside/firmware/include -I./quickassist/lookaside/access_layer/src/common/include -I./quickassist/lookaside/access_layer/include -I./quickassist/lookaside/access_layer/src/qat_direct/include -I./quickassist/lookaside/access_layer/src/qat_direct/common/include -I./quickassist/lookaside/access_layer/src/qat_direct/vfio -D USER_SPACE -D LAC_BYTE_ORDER=__LITTLE_ENDIAN -Wformat -Wformat-security -Werror=format-security -D_FORTIFY_SOURCE=2 -fstack-protector-strong -fPIE -fPIC -fno-strict-overflow -fno-delete-null-pointer-checks -fwrapv -D MAX_MR_ROUND=50 -D _FILE_OFFSET_BITS=64 -Wno-unused-command-line-argument -mcx16 -DICP_PARAM_CHECK -g -O2 -version-info 4:0:0 -pie -z relro -z now -z noexecstack -export-symbols-regex "^(cpa|icp_sal)" -o libqat.la -rpath /usr/local/lib quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_buffers.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_chain.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_datapath.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_dp.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_header_footer.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_header_footer_lz4.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_session.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_stats.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_err_sim.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_ns_datapath.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_ns_header_footer.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_crc32.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_crc64.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-dc_xxhash32.lo quickassist/lookaside/access_layer/src/common/compression/libqat_la-icp_sal_dc_err_sim.lo quickassist/lookaside/access_layer/src/common/crypto/asym/diffie_hellman/libqat_la-lac_dh_control_path.lo quickassist/lookaside/access_layer/src/common/crypto/asym/diffie_hellman/libqat_la-lac_dh_data_path.lo quickassist/lookaside/access_layer/src/common/crypto/asym/diffie_hellman/libqat_la-lac_dh_interface_check.lo quickassist/lookaside/access_layer/src/common/crypto/asym/diffie_hellman/libqat_la-lac_dh_stats.lo quickassist/lookaside/access_layer/src/common/crypto/asym/dsa/libqat_la-lac_dsa.lo quickassist/lookaside/access_layer/src/common/crypto/asym/dsa/libqat_la-lac_dsa_interface_check.lo quickassist/lookaside/access_layer/src/common/crypto/asym/ecc/libqat_la-lac_ec.lo quickassist/lookaside/access_layer/src/common/crypto/asym/ecc/libqat_la-lac_ec_common.lo quickassist/lookaside/access_layer/src/common/crypto/asym/ecc/libqat_la-lac_ec_montedwds.lo quickassist/lookaside/access_layer/src/common/crypto/asym/ecc/libqat_la-lac_ec_nist_curves.lo quickassist/lookaside/access_layer/src/common/crypto/asym/ecc/libqat_la-lac_ecdh.lo quickassist/lookaside/access_layer/src/common/crypto/asym/ecc/libqat_la-lac_ecdsa.lo quickassist/lookaside/access_layer/src/common/crypto/asym/ecc/libqat_la-lac_kpt_ecdsa.lo quickassist/lookaside/access_layer/src/common/crypto/asym/large_number/libqat_la-lac_ln.lo quickassist/lookaside/access_layer/src/common/crypto/asym/large_number/libqat_la-lac_ln_interface_check.lo quickassist/lookaside/access_layer/src/common/crypto/asym/pke_common/libqat_la-lac_pke_mmp.lo quickassist/lookaside/access_layer/src/common/crypto/asym/pke_common/libqat_la-lac_pke_qat_comms.lo quickassist/lookaside/access_layer/src/common/crypto/asym/pke_common/libqat_la-lac_pke_utils.lo quickassist/lookaside/access_layer/src/common/crypto/asym/prime/libqat_la-lac_prime.lo quickassist/lookaside/access_layer/src/common/crypto/asym/prime/libqat_la-lac_prime_interface_check.lo quickassist/lookaside/access_layer/src/common/crypto/asym/rsa/libqat_la-lac_rsa.lo quickassist/lookaside/access_layer/src/common/crypto/asym/rsa/libqat_la-lac_rsa_control_path.lo quickassist/lookaside/access_layer/src/common/crypto/asym/rsa/libqat_la-lac_rsa_decrypt.lo quickassist/lookaside/access_layer/src/common/crypto/asym/rsa/libqat_la-lac_rsa_encrypt.lo quickassist/lookaside/access_layer/src/common/crypto/asym/rsa/libqat_la-lac_rsa_interface_check.lo quickassist/lookaside/access_layer/src/common/crypto/asym/rsa/libqat_la-lac_rsa_keygen.lo quickassist/lookaside/access_layer/src/common/crypto/asym/rsa/libqat_la-lac_rsa_stats.lo quickassist/lookaside/access_layer/src/common/crypto/asym/rsa/libqat_la-lac_kpt_rsa_decrypt.lo quickassist/lookaside/access_layer/src/common/crypto/sym/drbg/libqat_la-lac_sym_drbg_api.lo quickassist/lookaside/access_layer/src/common/crypto/sym/key/libqat_la-lac_sym_key.lo quickassist/lookaside/access_layer/src/common/crypto/sym/libqat_la-lac_sym_alg_chain.lo quickassist/lookaside/access_layer/src/common/crypto/sym/libqat_la-lac_sym_api.lo quickassist/lookaside/access_layer/src/common/crypto/sym/libqat_la-lac_sym_auth_enc.lo quickassist/lookaside/access_layer/src/common/crypto/sym/libqat_la-lac_sym_cb.lo quickassist/lookaside/access_layer/src/common/crypto/sym/libqat_la-lac_sym_cipher.lo quickassist/lookaside/access_layer/src/common/crypto/sym/libqat_la-lac_sym_compile_check.lo quickassist/lookaside/access_layer/src/common/crypto/sym/libqat_la-lac_sym_dp.lo quickassist/lookaside/access_layer/src/common/crypto/sym/libqat_la-lac_sym_hash.lo quickassist/lookaside/access_layer/src/common/crypto/sym/libqat_la-lac_sym_partial.lo quickassist/lookaside/access_layer/src/common/crypto/sym/libqat_la-lac_sym_queue.lo quickassist/lookaside/access_layer/src/common/crypto/sym/libqat_la-lac_sym_stats.lo quickassist/lookaside/access_layer/src/common/crypto/sym/nrbg/libqat_la-lac_sym_nrbg_api.lo quickassist/lookaside/access_layer/src/common/crypto/sym/qat/libqat_la-lac_sym_qat.lo quickassist/lookaside/access_layer/src/common/crypto/sym/qat/libqat_la-lac_sym_qat_cipher.lo quickassist/lookaside/access_layer/src/common/crypto/sym/qat/libqat_la-lac_sym_qat_constants_table.lo quickassist/lookaside/access_layer/src/common/crypto/sym/qat/libqat_la-lac_sym_qat_hash.lo quickassist/lookaside/access_layer/src/common/crypto/sym/qat/libqat_la-lac_sym_qat_hash_defs_lookup.lo quickassist/lookaside/access_layer/src/common/crypto/sym/qat/libqat_la-lac_sym_qat_key.lo quickassist/lookaside/access_layer/src/common/crypto/sym/libqat_la-lac_sym_hash_sw_precomputes.lo quickassist/lookaside/access_layer/src/common/crypto/kpt/provision/libqat_la-lac_kpt_provision.lo quickassist/lookaside/access_layer/src/common/ctrl/libqat_la-sal_compression.lo quickassist/lookaside/access_layer/src/common/ctrl/libqat_la-sal_create_services.lo quickassist/lookaside/access_layer/src/common/ctrl/libqat_la-sal_ctrl_services.lo quickassist/lookaside/access_layer/src/common/ctrl/libqat_la-sal_list.lo quickassist/lookaside/access_layer/src/common/ctrl/ [... cut ...]
make[2]: Leaving directory '/home/kkurzacz/benchmark/qat/qatlib'
make[1]: Leaving directory '/home/kkurzacz/benchmark/qat/qatlib'
```

systemctl status:
```
$ systemctl status qat.service
× qat.service - QAT service
     Loaded: loaded (/lib/systemd/system/qat.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Fri 2023-11-03 14:28:25 UTC; 41s ago
        CPU: 604ms
```

journalctl does not return anything. What might be the problem?

### System info
Ubuntu jammy 22.04.3 LTS
gcc version 12.3.0 (Ubuntu 12.3.0-1ubuntu1~23.04)

jdschuet, 2023-11-20:
Can you the following details:

# sudo dmesg | grep qat
# sudo dmesg | grep vfio

kkurzacz-intel, 2023-12-05:
It happened that I had to switch the machine and redo all the installation process. It's possible I was missing some config files. Like /etc/4xxx for Saphire Rapids CPU. Let's close it for now.

closed 2023-12-05 by kkurzacz-intel