intel/qatlib #66: qat.service couldn't be found after installing with configure options  --enable-service

xiaochunlee, 2023-11-30:
Description of problem:
When run command "make install", an error message will appear and can not start qat.service after "./configure --enable-service" has been run

make install log:
Must remake target 'install-data-hook'.
make[2]: Entering directory '/root/qat/qatlib-main'
Putting child 0x5557b637ab40 (install-data-hook) PID 28921 on the chain.
Live child 0x5557b637ab40 (install-data-hook) PID 28921 
Failed to start qat.service: Unit qat.service not found.
Failed to enable unit: Unit file qat.service does not exist.
Reaping losing child 0x5557b637ab40 PID 28921 
make[2]: *** [Makefile:7140: install-data-hook] Error 1
Removing child 0x5557b637ab40 PID 28921 from chain.
make[2]: Leaving directory '/root/qat/qatlib-main'
Reaping losing child 0x55d30fa72980 PID 28919 
make[1]: *** [Makefile:6794: install-data-am] Error 2
Removing child 0x55d30fa72980 PID 28919 from chain.
make[1]: Leaving directory '/root/qat/qatlib-main'
Reaping losing child 0x55e91a1081c0 PID 28716 
make: *** [Makefile:6384: install-am] Error 2
Removing child 0x55e91a1081c0 PID 28716 from chain.
 
qat.service status:
root@t:~/qat/qatlib-main# systemctl status qat
Unit qat.service could not be found.

Version-Release number of selected component (if applicable):

ubuntu22.04.3  
qatlib version 23.11.0 
gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04) 
 
 
Steps to Reproduce:
1.  ./autogen.sh
2.  ./configure --enable-service
3.  make
4.  make install
 
Expect results:
command "make install" could automatically enable qat.service while using config command "./configure --enable-service"

Addtional info:
Add configure options systemdsystemunitdir can fix this issue, like ./configure --enable-service  **systemdsystemunitdir=/lib/systemd/system/**
Would it be possibe adding this as a note for different distros?

jdschuet, 2023-12-05:
Thanks for this feedback!  I'd like to ensure I understand the source of the issue...
Is the source of the problem that the install directory for qat.service unit not valid for Ubunutu distro?

xiaochunlee, 2023-12-06:
> Thanks for this feedback! I'd like to ensure I understand the source of the issue... Is the source of the problem that the install directory for qat.service unit not valid for Ubunutu distro?

Yes, I just did the source compile and installation on ubuntu22.04.3 and found the weird issue.

jdschuet, 2023-12-11:
Thanks for confirming.  We will update documentation accordingly.

jdschuet, 2024-03-28:
Details have been added in this section:
https://intel.github.io/quickassist/qatlib/install.html#installing-from-sources