intel/qatlib #68: Does KTLS support offloading crypto operation to QAT device?

liverbirdkte, 2023-12-20:
Hi guys, I'm looking at KTLS(https://docs.kernel.org/networking/tls-offload.html) which supports offloading crypto work to NICs. Is it possible to use QAT in this scenario, to let kernel offload crypto to QAT device? If so, do you have any performance benchmark? Thanks!

liverbirdkte, 2024-01-04:
Hi @jdschuet @fionatrahe, it seems KTLS crypto can only be offloaded to NIC hardware, there's no way to use QAT to accelerate encryption and decryption of kernel TLS. Is my understanding right?

gcabiddu, 2024-01-05:
Looking at the documentation and the code, it appears that kTLS can use accelerators through the Linux Crypto framework when running in Software crypto mode (TLS_SW).

liverbirdkte, 2024-01-09:
@gcabiddu, yes, it should work to specify qat related cipher with KTLS, I'll try with this, thanks!

closed 2024-01-09 by gcabiddu