intel/qatlib #69: Intel QAT service script issues and restriction

juggarnautss, 2024-01-15:
We are enabling QAT services on our platforms where we have to give configuration option to user to enable no of VF/endpoints for a specific QAT(PF) device. Once the endpoints are enabled on platform, we will using intel plugin operator to install the QAT plugin. However, on the platform side we found following restriction/constraints wrt intel /etc/init.d/qat_service script :

/etc/init.d/qat_service script does not support configuring user defined number of QAT VFs other than max supported ie. 16.

/etc/init.d/qat_service script does not support starting a specific QAT device, by default it starts all the devices irrespective of how many endpoint configuration files are present or not.

During multiple start or restart of single PF using intel qat_service script, mapping of device name with their configuration files is changing. 
Example: 
Initial VF mapping=> 
PF0 : qat_dev2(0000:f3:00.1) maps to  4xxxvf_dev0.conf  
PF1 : qat_dev18(0000:f7:00.1) maps to  4xxxvf_dev16.conf  
After restart PF0 or PF1=> 
PF0 : qat_dev2(0000:f7:00.1) maps to  4xxxvf_dev0.conf  
PF1 : qat_dev18(0000:f3:00.1) maps to  4xxxvf_dev16.conf  

Please assist us in resolving these limitation.

mythi, 2024-01-15:
Related answers https://github.com/intel/intel-device-plugins-for-kubernetes/issues/1646#issuecomment-1882589148

juggarnautss, 2024-01-18:
Hello All,

Any timeline or update on the limitation mentioned in the issue description.

Script location: 
https://github.com/intel/qatlib/tree/main/quickassist/utilities/service

jdschuet, 2024-01-22:
Hello,

Thanks for raising these questions.  I'm happy to report the limitation regarding missing control over number of VFs exposed per PF is not valid for the 4xxx device mentioned above. Details on controlling number of VFs exposed is included at: https://intel.github.io/quickassist/PG/virtualization.html?highlight=reducing#reducing-number-of-vfs-per-endpoint

With regards to non-deterministic mapping of the device IDs, can you share details on your Linux Distro and kernel version?

juggarnautss, 2024-02-01:
Hello @jdschuet ,

I have checked the link - https://intel.github.io/quickassist/PG/virtualization.html?highlight=reducing#reducing-number-of-vfs-per-endpoint.
Yes, we can limit the VF's per PF by using command "echo 4 > /sys/bus/pci/devices/0000\:6b:00.0/sriov_numvfs" , however in order to make the changes to take place we have to restart the qat_service.

Now, qat_service restart executes /etc/init.d/qat_service script which internally restarts all the VF's (QAT endpoints) and no of VF's again resets to default ie 16.

 Linux Distro and kernel version wrt  non-deterministic mapping of the device IDs:
Linux controller-0 5.10.0-6-rt-amd64 #1 SMP PREEMPT_RT StarlingX Debian 5.10.205-1.stx.76 (2024-01-24 x86_64 GNU/Linux

jdschuet, 2024-02-14:
Hi @juggarnautss,

The information above corresponds with our Out-of-Tree (OOT) QAT kernel modules that are available at:
https://www.intel.com/content/www/us/en/download/765501.html.    Currently the in-tree QAT kernel modules have not been updated with this capability. As such we are updating the release notes for the in-tree QAT kernel modules at: https://intel.github.io/quickassist/RN/Linux/2.X/updates.html#release-updates with this limitation.

Given the details above, it looks like you are able to reduce the number of VFs, but in course of restarting the driver to reflect these changes, all VFs are populated?  This is different behavior than what was observed with CentOS.  Based on the about above, this is observed with Debian based distro?  I'd like to create the issue locally.

Thanks,
Joel

jdschuet, 2024-02-15:
Hi @juggarnautss,

I've recreated the issue on Ubuntu 22.04.  Next step is to understand why behavior is different here compared to other distros.

Thanks,
Joel

jdschuet, 2024-07-01:
Hi @juggarnautss,

Documentation has been updated to reflect usage.  In short restart of QAT driver is not required.   Documentation also notes that if restart of driver is done, we revert to the full set of VFs.  https://intel.github.io/quickassist/PG/virtualization.html#reducing-number-of-vfs-per-endpoint

Thanks,
Joel

fionatrahe, 2024-07-08:
closing this as resolved

closed 2024-07-08 by fionatrahe