intel/qatlib #77: Does qatlib support hash-then-compress?

neel-patel-1, 2024-03-11:
When enabling dcc in the qat configuration file, sample hash-then-compress sample code shows that hash-then-compress operations are not supported:

Is there a way to configure use of hash-then-compress?

```
(base) n869p538@sapphire:qatlib$ cat /etc/sysconfig/qat
# Comment or remove next line to disable sriov
#SRIOV_ENABLE=1
POLICY=16
ServicesEnabled=dcc
(base) n869p538@sapphire:qatlib$ sudo systemctl restart qat
(base) n869p538@sapphire:qatlib$ ./cpa_sample_code runTests=128
Latency computation Disabled
qaeMemInit started
icp_sal_userStartMultiProcess("SSL") started
There are no crypto instances
*** QA version information ***
device ID               = 0
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 1
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 2
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 3
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 4
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 5
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 6
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 7
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 8
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 9
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 10
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 11
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 12
software                = 24.2.0
*** END QA version information ***
*** QA version information ***
device ID               = 13
[... 115 lines cut ...]
Hash + compress chained operation is not supported on logical instance 28
Hash + compress chained operation is not supported on logical instance 24
Hash + compress chained operation is not supported on logical instance 58
Hash + compress chained operation is not supported on logical instance 11
Hash + compress chained operation is not supported on logical instance 32
Hash + compress chained operation is not supported on logical instance 26
Hash + compress chained operation is not supported on logical instance 34
Hash + compress chained operation is not supported on logical instance 40
Hash + compress chained operation is not supported on logical instance 43
Hash + compress chained operation is not supported on logical instance 33
Hash + compress chained operation is not supported on logical instance 13
Hash + compress chained operation is not supported on logical instance 8
Hash + compress chained operation is not supported on logical instance 16
Hash + compress chained operation is not supported on logical instance 19
Hash + compress chained operation is not supported on logical instance 22
Hash + compress chained operation is not supported on logical instance 38
Hash + compress chained operation is not supported on logical instance 31
Hash + compress chained operation is not supported on logical instance 45
Hash + compress chained operation is not supported on logical instance 25
Hash + compress chained operation is not supported on logical instance 42
Hash + compress chained operation is not supported on logical instance 3
Hash + compress chained operation is not supported on logical instance 4
Hash + compress chained operation is not supported on logical instance 44
---------------------------------------
```

neel-patel-1, 2024-03-11:
Our system is running Linux 6.5.7

Is a newer version required for this feature?

neel-patel-1, 2024-03-15:
Having difficulties enabling chaining on kernel 6.7:
```
(base) n869p538@sapphire:qatlib$ cat /etc/sysconfig/qat
# Comment or remove next line to disable sriov
#SRIOV_ENABLE=1
POLICY=1
ServicesEnabled=dcc
(base) n869p538@sapphire:qatlib$ uname -r
6.7.0-060700-generic
(base) n869p538@sapphire:interrupt_qat_dc$ sudo systemctl status qat.service
× qat.service - QAT service
     Loaded: loaded (/lib/systemd/system/qat.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2024-03-15 15:46:34 CDT; 38s ago
TriggeredBy: ● qat.timer
    Process: 164468 ExecStartPre=/bin/sh -c test $(getent group qat) (code=exited, status=0/SUCCESS)
    Process: 164475 ExecStartPre=/usr/local/sbin/qat_init.sh (code=exited, status=0/SUCCESS)
    Process: 164763 ExecStart=/usr/local/sbin/qatmgr --policy=${POLICY} (code=exited, status=1/FAILURE)
        CPU: 1.020s

Mar 15 15:46:33 sapphire.ittc.ku.edu qat_init.sh[164475]: /usr/local/sbin/qat_init.sh, device 0000:76:00.0 configured with services: dcc
Mar 15 15:46:33 sapphire.ittc.ku.edu qat_init.sh[164475]: /usr/local/sbin/qat_init.sh, device 0000:f3:00.0 configured with services: dcc
Mar 15 15:46:33 sapphire.ittc.ku.edu qat_init.sh[164475]: /usr/local/sbin/qat_init.sh: 364: echo: echo: I/O error
Mar 15 15:46:34 sapphire.ittc.ku.edu qat_init.sh[164475]: /usr/local/sbin/qat_init.sh: 364: echo: echo: I/O error
Mar 15 15:46:34 sapphire.ittc.ku.edu qatmgr[164765]: No devices found
Mar 15 15:46:34 sapphire.ittc.ku.edu qatmgr[164763]: No QAT device found
Mar 15 15:46:34 sapphire.ittc.ku.edu systemd[1]: qat.service: Control process exited, code=exited, status=1/FAILURE
Mar 15 15:46:34 sapphire.ittc.ku.edu systemd[1]: qat.service: Failed with result 'exit-code'.
Mar 15 15:46:34 sapphire.ittc.ku.edu systemd[1]: Failed to start QAT service.
Mar 15 15:46:34 sapphire.ittc.ku.edu systemd[1]: qat.service: Consumed 1.020s CPU time.
```

Starting with 
ServicesEnabled=dc
```
● qat.service - QAT service
     Loaded: loaded (/lib/systemd/system/qat.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2024-03-15 15:48:03 CDT; 3s ago
TriggeredBy: ● qat.timer
    Process: 165277 ExecStartPre=/bin/sh -c test $(getent group qat) (code=exited, status=0/SUCCESS)
    Process: 165279 ExecStartPre=/usr/local/sbin/qat_init.sh (code=exited, status=0/SUCCESS)
    Process: 165931 ExecStart=/usr/local/sbin/qatmgr --policy=${POLICY} (code=exited, status=0/SUCCESS)
   Main PID: 165933 (qatmgr)
      Tasks: 1 (limit: 308979)
     Memory: 2.3M
        CPU: 1.748s
     CGroup: /system.slice/qat.service
             └─165933 /usr/local/sbin/qatmgr --policy=1

Mar 15 15:48:03 sapphire.ittc.ku.edu qatmgr[165933]: Detected DC configuration
Mar 15 15:48:03 sapphire.ittc.ku.edu qatmgr[165933]: section 8, BDF 7602
Mar 15 15:48:03 sapphire.ittc.ku.edu qatmgr[165933]: Detected DC configuration
Mar 15 15:48:03 sapphire.ittc.ku.edu qatmgr[165933]: section 9, BDF F302
Mar 15 15:48:03 sapphire.ittc.ku.edu qatmgr[165933]: Detected DC configuration
Mar 15 15:48:03 sapphire.ittc.ku.edu qatmgr[165933]: section 10, BDF 760A
Mar 15 15:48:03 sapphire.ittc.ku.edu qatmgr[165933]: Detected DC configuration
Mar 15 15:48:03 sapphire.ittc.ku.edu qatmgr[165933]: section 11, BDF F30A
Mar 15 15:48:03 sapphire.ittc.ku.edu systemd[1]: Started QAT service.
Mar 15 15:48:03 sapphire.ittc.ku.edu qatmgr[165933]: Detected DC configuration
```

Do you know if there is a configuration option I am missing?

gcabiddu, 2024-03-15:
Hi @neel-patel-1, can you confirm you have the latest firmware image from https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/qat_4xxx.bin

If not, can you copy it to /lib/firmware, reload the qat driver, restart the qat service and optionally rebuild the initramfs?
```
sudo mv qat_4xxx.bin /lib/firmware
sudo rmmod qat_4xxx intel_qat
sudo modprobe qat_4xxx
systemctl restart qat
dracut -f # if you are using fedora
```

neel-patel-1, 2024-03-15:
```
(base) n869p538@sapphire:interrupt_qat_dc$ sudo mv qat_4xxx.bin /lib/firmware
(base) n869p538@sapphire:interrupt_qat_dc$ sudo rmmod qat_4xxx intel_qat
(base) n869p538@sapphire:interrupt_qat_dc$ sudo modprobe qat_4xxx
(base) n869p538@sapphire:interrupt_qat_dc$ systemctl restart qat
```
Yes these steps worked:
```
(base) n869p538@sapphire:interrupt_qat_dc$ sudo systemctl status qat.service
● qat.service - QAT service
     Loaded: loaded (/lib/systemd/system/qat.service; enabled; vendor preset: enabled)
     Active: active (running) since Fri 2024-03-15 15:57:26 CDT; 39s ago
TriggeredBy: ● qat.timer
    Process: 167094 ExecStartPre=/bin/sh -c test $(getent group qat) (code=exited, status=0/SUCCESS)
    Process: 167096 ExecStartPre=/usr/local/sbin/qat_init.sh (code=exited, status=0/SUCCESS)
    Process: 167806 ExecStart=/usr/local/sbin/qatmgr --policy=${POLICY} (code=exited, status=0/SUCCESS)
   Main PID: 167808 (qatmgr)
      Tasks: 1 (limit: 308979)
     Memory: 1.8M
        CPU: 1.964s
     CGroup: /system.slice/qat.service
             └─167808 /usr/local/sbin/qatmgr --policy=1

Mar 15 15:57:26 sapphire.ittc.ku.edu qatmgr[167808]: Detected DC configuration
Mar 15 15:57:26 sapphire.ittc.ku.edu qatmgr[167808]: section 8, BDF 7602
Mar 15 15:57:26 sapphire.ittc.ku.edu qatmgr[167808]: Detected DC configuration
Mar 15 15:57:26 sapphire.ittc.ku.edu qatmgr[167808]: section 9, BDF F302
Mar 15 15:57:26 sapphire.ittc.ku.edu qatmgr[167808]: Detected DC configuration
Mar 15 15:57:26 sapphire.ittc.ku.edu qatmgr[167808]: section 10, BDF 760A
Mar 15 15:57:26 sapphire.ittc.ku.edu qatmgr[167808]: Detected DC configuration
Mar 15 15:57:26 sapphire.ittc.ku.edu qatmgr[167808]: section 11, BDF F30A
Mar 15 15:57:26 sapphire.ittc.ku.edu systemd[1]: Started QAT service.
Mar 15 15:57:26 sapphire.ittc.ku.edu qatmgr[167808]: Detected DC configuration
```

Thank you for your help

closed 2024-03-15 by neel-patel-1