intel/qatlib #78: Will the c6xx series be considered for support?

weeweetan, 2024-03-27:


fionatrahe, 2024-04-15:
Since the beginning, qatlib has been designed to leverage the advanced features of hardware from generation 4 and beyond! As a result, supporting c6xx series is not a possibility. See  supported devices here: [https://intel.github.io/quickassist/qatlib/requirements.html#supported-devices](url)

sfiros, 2024-04-15:
The URL @fionatrahe gives the full clarity. I presume @weeweetan had the question by not particularly seeing the c6xx in the not-supported list in the README (we had c62x, dh895xxcc, etc. in the line under the section 'Supported Devices') but that was provided as some examples of it and not as the full list, but I think it could be a good idea to update the README such that c6xx will be added in place of c62x which will cover the whole series of it that are not supported. We will update the README in the next release if that helps.