intel/qatlib #80: DataPlane API Overflow Recovery

achains, 2024-05-03:
Hello! 

Some of your documents states that overflow on compression using DataPlane API is unrecoverable error. 

Could you please specify what exactly does it mean? Can't I just catch the error code in callback function and then re-submit the right request? What will happen to other requests submitted in one batch if one of them fails with overflow? 

Thank you in advance!

gcabiddu, 2024-05-07:
Hi @achains,
> Could you please specify what exactly does it mean?

if you get an overflow condition you need to resubmit the job increasing the size of the destination buffer.

> Can't I just catch the error code in callback function and then re-submit the right request?

Yes, you can

> What will happen to other requests submitted in one batch if one of them fails with overflow?

Other requests are not impacted. Are you seeing a different behaviour?

achains, 2024-05-07:
Thank you for your answer. I was just confused by a word "Unrecoverable". In case of traditional API the driver doesn't log this "unrecoverable" error. So I was wondering whether resubmitting a request can lead to undefined behavior with DataPlane API. 

Here is a paragraph from "Programmers Guide: Hardware Version 2.0"   
[image]

fionatrahe, 2024-10-14:
thanks for your feedback. We plan to change the text to avoid mentioning Unrecoverable in a future release.

closed 2024-10-14 by fionatrahe