intel/qatlib #82: a code security scanner reports a number of warnings for qatlib code

nefigtut, 2024-06-25:
hi, openscanhub code security scanner reports a number of Missing Initialization and Resource Leaks in the qatlib code. they can be a low-impact, but still i would like to share security scanner findings. thanks!

[scan-results-imp.tar.gz](https://github.com/user-attachments/files/15972728/scan-results-imp.tar.gz)

fionatrahe, 2024-06-25:
thanks! we'll investigate.

fionatrahe, 2024-10-14:
@nefigtut Can you please rerun openscanhub with 24.09 and let us know if all issues were resolved.

nefigtut, 2024-10-22:
Hi, Fiona,
I've scanned 24.09 using the previous scan of 24.02 as a base. All the important issues were fixed except one. Thank you and your team! The results archive is attached, could you please have a look?

[scan-results-imp-24-09.tar.gz](https://github.com/user-attachments/files/17481803/scan-results-imp-24-09.tar.gz)

fionatrahe, 2024-12-16:
Thanks @nefigtut, we're investigating this case and aim to fix it in the next release.