intel/qatlib #100: reserved identifier violation

elfring, 2025-01-24:
I would like to point out that an identifier like “[`_CpaDeviceInfo`](https://github.com/intel/qatlib/blob/240991f2655fb2f889715003b22dfbba89e9b90f/quickassist/include/cpa_dev.h#L110 "Update candidate")” [does eventually not fit](https://wiki.sei.cmu.edu/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

Ahsan-Atta, 2025-01-30:
Hi Markus,

This could have implications for maintaining API compliance and we have requirements that do not allow us to break that.

To what level and severity are you impacted by this? 

Thanks
Ahsan

elfring, 2025-01-30:
* :eyes: Please take another look at [name conventions](https://stackoverflow.com/questions/228783/what-are-the-rules-about-using-an-underscore-in-a-c-identifier#answer-228797 "What are the rules about using an underscore in a C++ identifier?").
  How do you think about to avoid that this software [depends on undefined behaviour](https://en.cppreference.com/w/cpp/language/identifiers#In_declarations "Documentation")? :thinking:
* :thought_balloon: Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/bugprone/reserved-identifier.html "Checking the application of reserved identifiers")?

Ahsan-Atta, 2026-02-09:
I believe this is resolved in QATlib 26.02. Please let me know if anything else needs to be changed.