intel/qatlib #101: Completion of error handling

elfring, 2025-01-24:
Would you like to add more error handling for return values from functions like the following?
* [fclose](https://pubs.opengroup.org/onlinepubs/9799919799/functions/fclose.html "Close a stream.") ⇒ [write_pidfile](https://github.com/intel/qatlib/blob/240991f2655fb2f889715003b22dfbba89e9b90f/quickassist/utilities/qat_mgr/qat_mgr.c#L278)
* [fprintf](https://pubs.opengroup.org/onlinepubs/9799919799/functions/fprintf.html "Print formatted output.") ⇒ [osalLog](https://github.com/intel/qatlib/blob/240991f2655fb2f889715003b22dfbba89e9b90f/quickassist/utilities/osal/src/linux/user_space/OsalServices.c#L134-L138)

Ahsan-Atta, 2025-01-30:
Hi Markus,

Thanks for this. We'll investigate this and get back to you.

Regards,
Ahsan

elfring, 2025-01-30:
How do you think about to improve [static source code analysis](https://en.wikipedia.org/wiki/List_of_tools_for_static_code_analysis#C.2C_C.2B.2B "Tools for code analysis") also for this software? :thinking:

Ahsan-Atta, 2025-07-17:
Resolved in QATlib 25.08 release. Hence, closing this issue.

elfring, 2025-07-18:
Thanks for some software improvements.

:thinking:
* [Can memory allocations be occasionally questionable for error reporting](https://github.com/intel/qatlib/blob/f867a08fd279548a953b9f0b631842ae83a351e9/quickassist/utilities/qat_mgr/qat_mgr.c#L287-L295 "Update candidate: write_pidfile()")?
* Would you become interested in [the means of aspect-oriented software development](https://en.wikipedia.org/wiki/Aspect-oriented_programming "Documentation")?