intel/qatlib #102: QAT Service fails to start

akarX23, 2025-02-06:
When I install the drivers using the instructions mentioned in the [getting started guide](https://intel.github.io/quickassist/GSG/2.X/installation.html#installation-overview), everything seems to be working fine. But I want the upstreamed acceleration software.

I am following the [guide here](https://intel.github.io/quickassist/qatlib/install.html), however, after the `./configure` step, I am getting this error:
```
make[2]: Leaving directory '/root/qatlib'
make[1]: Leaving directory '/root/qatlib'
[root@pocmasternode qatlib]# sudo systemctl enable qat
[root@pocmasternode qatlib]# sudo systemctl enable qat
Created symlink /etc/systemd/system/multi-user.target.wants/qat.service → /usr/lib/systemd/system/qat.service.
[root@pocmasternode qatlib]# sudo systemctl start qat
Job for qat.service failed because the control process exited with error code.
See "systemctl status qat.service" and "journalctl -xeu qat.service" for details.
[root@pocmasternode qatlib]# systemctl status qat
× qat.service - QAT service
     Loaded: loaded (/usr/lib/systemd/system/qat.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Thu 2025-02-06 11:07:08 EST; 9s ago
    Process: 190046 ExecStartPre=/bin/sh -c test $(getent group qat) (code=exited, status=0/SUCCESS)
    Process: 190048 ExecStartPre=/usr/local/sbin/qat_init.sh (code=exited, status=1/FAILURE)
        CPU: 1.415s

Feb 06 11:06:47 pocmasternode systemd[1]: Starting QAT service...
Feb 06 11:07:08 pocmasternode qat_init.sh[190048]: QAT driver is still not present after 20.0s. Aborting qat_init
Feb 06 11:07:08 pocmasternode systemd[1]: qat.service: Control process exited, code=exited, status=1/FAILURE
Feb 06 11:07:08 pocmasternode systemd[1]: qat.service: Failed with result 'exit-code'.
Feb 06 11:07:08 pocmasternode systemd[1]: Failed to start QAT service.
Feb 06 11:07:08 pocmasternode systemd[1]: qat.service: Consumed 1.415s CPU time.
```

The service status says that `QAT driver is still not present after 20.0s`, but I am sure that QAT is definitely present. When following the getting started guide, this is the output:
```
[root@pocmasternode QAT]# sudo systemctl status qat
● qat.service - QAT service
     Loaded: loaded (/usr/lib/systemd/system/qat.service; enabled; preset: disabled)
     Active: active (exited) since Thu 2025-02-06 10:57:14 EST; 51s ago
    Process: 172461 ExecStart=/etc/init.d/qat_service start (code=exited, status=0/SUCCESS)
   Main PID: 172461 (code=exited, status=0/SUCCESS)
        CPU: 2.883s

Feb 06 10:57:14 pocmasternode qat_service[172596]: There is 8 QAT acceleration device(s) in the system:
Feb 06 10:57:14 pocmasternode qat_service[172596]:  qat_dev0 - type: 4xxx,  inst_id: 0,  node_id: 1,  bsf: 0000:01:00.0,  #accel: 1 #engines: 9 state: up
Feb 06 10:57:14 pocmasternode qat_service[172596]:  qat_dev1 - type: 4xxx,  inst_id: 1,  node_id: 1,  bsf: 0000:06:00.0,  #accel: 1 #engines: 9 state: up
Feb 06 10:57:14 pocmasternode qat_service[172596]:  qat_dev2 - type: 4xxx,  inst_id: 2,  node_id: 1,  bsf: 0000:0b:00.0,  #accel: 1 #engines: 9 state: up
Feb 06 10:57:14 pocmasternode qat_service[172596]:  qat_dev3 - type: 4xxx,  inst_id: 3,  node_id: 1,  bsf: 0000:10:00.0,  #accel: 1 #engines: 9 state: up
Feb 06 10:57:14 pocmasternode qat_service[172596]:  qat_dev4 - type: 4xxx,  inst_id: 4,  node_id: 4,  bsf: 0000:81:00.0,  #accel: 1 #engines: 9 state: up
Feb 06 10:57:14 pocmasternode qat_service[172596]:  qat_dev5 - type: 4xxx,  inst_id: 5,  node_id: 4,  bsf: 0000:86:00.0,  #accel: 1 #engines: 9 state: up
Feb 06 10:57:14 pocmasternode qat_service[172596]:  qat_dev6 - type: 4xxx,  inst_id: 6,  node_id: 4,  bsf: 0000:8b:00.0,  #accel: 1 #engines: 9 state: up
Feb 06 10:57:14 pocmasternode qat_service[172596]:  qat_dev7 - type: 4xxx,  inst_id: 7,  node_id: 4,  bsf: 0000:90:00.0,  #accel: 1 #engines: 9 state: up
Feb 06 10:57:14 pocmasternode systemd[1]: Finished QAT service.
```

Is there something I am doing wrong when using the upstream acceleration software? My CPU is a 120-core GNR AP. Please let me know if I can workaround this issue. Thanks!

fionatrahe, 2025-02-06:
From the logs above it seems you are using an out-of-tree package, e.g. dd you get it from here?
 https://www.intel.com/content/www/us/en/download/765501/intel-quickassist-technology-driver-for-linux-hw-version-2-0.html

If you want to use the in-tree stack for QAT, i.e. the linux kernel in-tree driver with the qatlib package in this repo, then please follow the instructions here: https://intel.github.io/quickassist/qatlib/index.html

If you have any further issues, get back to us here.

akarX23, 2025-02-06:
@fionatrahe thank you for replying.

I tried both the approaches and the out-of-tree approach did work for me.

But going forward, I believe the in-tree stack is more preferred. I have already tried the installation instructions mentioned here:
https://intel.github.io/quickassist/qatlib/index.html

However, I face an issue after the './configure' step. The qat_service fails to start.
Output of `systemctl status qat_service`:
```
[root@pocmasternode qatlib]# systemctl status qat
× qat.service - QAT service
     Loaded: loaded (/usr/lib/systemd/system/qat.service; enabled; preset: disabled)
     Active: failed (Result: exit-code) since Thu 2025-02-06 11:07:08 EST; 9s ago
    Process: 190046 ExecStartPre=/bin/sh -c test $(getent group qat) (code=exited, status=0/SUCCESS)
    Process: 190048 ExecStartPre=/usr/local/sbin/qat_init.sh (code=exited, status=1/FAILURE)
        CPU: 1.415s

Feb 06 11:06:47 pocmasternode systemd[1]: Starting QAT service...
Feb 06 11:07:08 pocmasternode qat_init.sh[190048]: QAT driver is still not present after 20.0s. Aborting qat_init
Feb 06 11:07:08 pocmasternode systemd[1]: qat.service: Control process exited, code=exited, status=1/FAILURE
Feb 06 11:07:08 pocmasternode systemd[1]: qat.service: Failed with result 'exit-code'.
Feb 06 11:07:08 pocmasternode systemd[1]: Failed to start QAT service.
Feb 06 11:07:08 pocmasternode systemd[1]: qat.service: Consumed 1.415s CPU time.
```

Is this a known issue? If not, I will put a list of all commands I executed and the environment details. Thank you.

gcabiddu, 2025-02-06:
Hi @akarX23, can you report the output of dmesg?

gcabiddu, 2025-02-06:
Yes, uninstall the QAT out-of-tree driver, please. The qat_init.sh script in qatlib is looking at an attribute in sysfs exposed by the QAT in-tree driver, and then aborting as it cannot find it.

Also, which distro are you using?

gcabiddu, 2025-02-06:
Can you now load the QAT upstream driver? `modprobe qat_4xxx` and then re-install qatlib?

akarX23, 2025-02-06:
`modprobe qat_4xxx` gives an error:
`modprobe: FATAL: Module qat_4xxx not found in directory /lib/modules/6.6.0-srf.bkc.6.6.20.6.19.x86_64`

Should I execute qatlib installation instructions first and then do this?

akarX23, 2025-02-10:
Thanks @gcabiddu for providing the solution - The kernel needs to be compiled with the QAT Modules as mentioned here:

https://intel.github.io/quickassist/qatlib/requirements.html#kernel-drivers

closed 2025-02-10 by akarX23